IanSB/RGBtoHDMI #3: amiga hoglet, software rgb/hdmi

c1581, 2021-06-23:
First time on Github, so forgive me if this is not the correct way to ask a question.

I have the rgm/hdmi adapter, commodore amiga 500, with an ACA500 plus.   Installed the adapter, on Pi0 checking for proper orientation,  software on SD card is:  20210529_777ea4f Release.

I get the "four color screen",  Amiga boots to the aca500 menu..   I get a blank screen, unless I press the spacebar, then I get a picture..  Booting, into the Amiga, screen goes blank and stays blank..   Removing the aca500 card, and just booting the amiga "plain", with 1.3 or 3.1 and again, just get a blank screen.

Is there some configuration I am missing?   I've tried it with two different Pi0's,  two different  Tv's

Thank you.

IanSB, 2021-06-28:
Which type of RGBtoHDMI board are you using?
Can you call up the RGBtoHDMI menus?

c1581, 2021-06-28:
I figured it out. I wasn’t getting any menus or anything. I soldered momentary push button on it and was able to determine it was the hertz setting. Increased from 50 to 60 and it was fine.

Thank you for getting back to me

Sean

Sent with [ProtonMail](https://protonmail.com/) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

closed 2021-06-30 by IanSB