IanSB/RGBtoHDMI #9: Is the buffer board needed only on the Atari ST?

mikrosk, 2022-03-05:
After reading this: https://github.com/c0pperdragon/Amiga-Digital-Video/issues/6#issuecomment-766461800 and this: https://github.com/c0pperdragon/Amiga-Digital-Video/issues/56#issuecomment-933144980 it would seem that STE models do not really require this board? Not that it would be easily possible to install it (DIP vs QFP package of later Shifter models) but still it makes me wondering whether ST signal is really that bad and much improved in STE models?

IanSB, 2022-03-08:
No, the buffer board isn't needed in the STE. It isn't really needed in the ST either as you can solder directly from the 12 bit extender cable to the shifter pins but it does make fitting it easier as the upgrade is mostly plugin. I added the buffer chips to try to protect the shifter from any damage (e.g. static discharge through the cable) and they aren't really needed for signal quality reasons.

mikrosk, 2022-03-08:
Thanks. While we are on the topic, where do the three wires go?

IanSB, 2022-03-08:
They are HSYNC, VSYNC and BLANK and they are available on the Glue chip, details in the "Atari ST 40 pin shifter" section of the cables guide in the wiki:
https://github.com/IanSB/RGBtoHDMI/wiki/Cables
The H&VSYNC signals are also available on the modulator header (be careful when connecting as there is also 12V on that header) and the BLANK signal is also available on diodes around the shifter chip but the actual component numbers differ depending on the ST variant so you will need to check the schematic.

mikrosk, 2022-03-08:
Much appreciated, @IanSB.

closed 2022-03-08 by mikrosk