IanSB/RGBtoHDMI #11: External Amiga Connection possible?

thbreidenbach, 2022-04-02:
The internal Amiga CPLD Boards are great but I do a lot of board repairs where initial testing is easier with an external Adapter like the D520 (which is unobtainium as of now) 
I wonder how the chances are to use the external 12bit Board on the Video Port of an Amiga? Even on AGA Machines I had great success with the internal Adapter even though there's missing colors in higher color depth. For testing or Workbench this is a non issue.

IanSB, 2022-04-02:
The external adapter should work with the Amiga's digital output on the Video Port but as it's only RGBI there will be limited colours. No need for the 12 bit extender as the bits would connect to the standard 8 bit header much like a CGA PC.

thbreidenbach, 2022-04-02:
Hi, thanks for your answer. Isn’t digital out (rgbd) more like the digital representation of composite (b/w) output?

IanSB, 2022-04-03:
>Isn’t digital out (rgbd) more like the digital representation of composite (b/w) output

I don't know, I've not tried it but the bits are labelled Digital R, G, B and Intensity which is the same as IBM CGA which gives 16 possible colours. The mapping of which digital bits are set depending on the input 12 bit RGB is done in the hybrid and I'm guessing it is mapping the input colour to the nearest RGBI colour for a limited palette.