IanSB/RGBtoHDMI #12: Atari STE - 9-bit adopted from STE (12-bit will come using Amiga pickup)

luciust, 2022-05-28:
Hi, thanks to @IanSB guidence - I've adopted temprorary Atari ST pickup board intended for Shifter (40-pin) to Atari STE.
The process requirer soldering and will only use 9-bits - not 12-bits, but thanks to the pickup board and bus transcievers - it makes your Atari STE safe from electrical mishaps.
Closeup of the actuall solders:

(Yellow cable is MONO)
[image]
HSYNC - right pin & VSYNC on left pin):
[image]

Results:
Mono is of course perfect:
[image]
Colour - it has some tint - i.e. white has cyan tint, red is a bit wahsed out, but it's far beter then expected:
[image]
[image]

Let me know if this need more explenation in meantime - I'm waiting for Amiga pickup that will add the 12-bit mode. Also - soldering will always be required, as the clearance for the signals pickup is HARD - my STE has PSU on top of it.

luciust, 2022-06-12:
@IanSB 
So I've used the Amiga 12-bit pickup board, tested color modes:
- The HSYNC and VSYNC signals have been replaced by Composite Sync from PIN 2 of MC1377 (Color Television RGB to PAL/NTSC Encoder) which is sitting before modulator
- using your 12 bit adapter for RGB2HDMI issue 2 
- Mono signal has been attached to PIN of U3 as suggested by you and the chips leg is lifted
- Beta54 FW and STE profile
- RPi Zero and Zero W (testing on two different units)

The outcome:

- in game menu - the picture is exactly *the same* as with 9-bit signals (0-2 mode) which I tested before - it's darker than original - with colors close to real output - but the colors are completly screwed (dakr green is pink and so on!) - but just during the actual game of Lotus (STE version) - blitter is running then.
- in GEM mode, the while is now white (without cyan tint) and the colors in the GEM menu are OK, but only first 8, where the next 8 are same color as 8 before. On regular monitor - the next 8 colors are lighter - which coresponds with above "darker" picture.

Other:
Minimal color artefacts are visible on some of the parts of picture,

Wondering what might be the issue?

IanSB, 2022-06-13:
First check that the sampling menu is set to 12bits per pixel with the STE profile.
Then check that the bits are in the right order
i.e. G3 on STE is connected to G3 on the Amiga pickup board, G2 <> G2, G1<> G1, G0<>G0 etc same for red & blue

IanSB, 2022-06-13:
Also need to check your soldering for shorts or unconnected pins

IanSB, 2022-06-13:
>I still haven't double-checked if my info on the R-G-B signals is correct and correctly wired - so that is still to do

That seems to be the likely cause. If the STE is anything like the ST you need to get the exact schematic for your motherboard because the different revisions of the ST all had different numbering for the resistors so using someone else's info might not work.

The grn1/VS pin will have the mono signal in mono mode but should be inactive in colour modes.

Is there a test pattern program for the STE?
On the Amiga I use the "Amiga Test Kit" and that displays ramps of RGB which makes it immediately obvious which bits are wrong:
[image]

luciust, 2022-06-14:
@IanSB The actual board - the MONO hack uses unused socket to meet the patch with mono signal (white cable), the purple is the CSYNC, and R G B are of course RGBs...
[image]

IanSB, 2022-06-14:
@luciust 

>Thanks to Lopez from local Discord (greetz!) I got the gray tester.
>Look - the 1-3 or 12-bits are the same wrong. But the mode 0-2 is nearly OK, look:

That confirms that you have all the bits rotated by 1
Note that the ST (not STE) is unusual in that you connect the high source bits to R2, G2, B2 and leave R3, G3, B3 used for other things (hence "9 bits (bits 0-2)"). 
The STE has the more standard connections where the highest bits go to R3,G3,B3.
The STE should look correct when using "9 bits (bits 1-3)" and wrong when using "9 bits (bits 0-2)" but your results are the other way around, hence you have connected STE RGB3 to input RGB2, STE RGB2 to input RGB1 and STE RGB1 to input RGB0.
When you enable full 12 bits per pixel you get alternate dark and light stripes because STE RGB0 is connected to input RGB3 (the brightest part of the signal which alternates on/off with every bar)

So you need to rotate all your RGB connections by 1

luciust, 2022-06-14:
@IanSB 
That seems to be right, and there's another issue in the C0pperDragon forum.

That also means that the STE Shifter datasheet I got is probably wrong. 

Anyway, when I get back I hope to nail it finally, and when sucessfull I'll share the details in FAQ - since I've broken pretty much everything by noe, and found 4 different data on the R,G,B signals :)

Thanks for patience!

luciust, 2022-06-14:
For reference: this is the GST Shifter "datasheet" I've been checking my signals with: 
[image]

luciust, 2022-06-15:
Finally - it works!

Based on your suggestions and pixelk0 schema - it works very well:
https://user-images.githubusercontent.com/5732483/124193866-e2011180-dac7-11eb-95ec-914855c7fa58.jpg

[image]

IanSB, 2022-06-15:
@luciust 

Does mono mode work as well?

IanSB, 2022-06-16:
The genlock is inhibited by default in mono mode because some monitors can't lock above 60Hz and mono mode is 71 Hz. Inhibiting the genlock forces the HDMI output to 60Hz which means you should always get an image on the screen. 
If your monitor can lock to 71Hz you can try changing the refresh to "Force 50Hz-Any" which should allow the HDMI output to always track the frequency of the input. 
Use menu+reset to recover if this doesn't work.

luciust, 2022-06-16:
Good idea, I got two more make-SHIFT HDMI monitors - fun thing to try!