IanSB/RGBtoHDMI #13: Amiga 500 -colors are displayed incorrectly | black is dark red

sascha70, 2022-06-05:
Hello community,

unfortunately a bug is driving me crazy with the Amiga 500. 

Unfortunately, the colors are displayed somewhat incorrectly for me. What should be black is dark red. I used the Beta54 version from JanBeta and the version from main tree 20211224_c97f06a release. All have the same problem. The adapter is from LinuxJedi (https://github.com/LinuxJedi/AmigaRGBtoHDMI/tree/main/Amiga500CPLD).

Do you have a solution?

Thanks

sascha
[image]
a

solarmon, 2022-06-05:
Looks like one of the red signal is missing or shorted together? I would inspect and check the soldering of the red signals and traces on the adapter itself.

Did you build the adapter yourself?

sascha70, 2022-06-05:
> Looks like one of the red signal is missing or shorted together? I would inspect and check the soldering of the red signals and traces on the adapter itself.
> 
> Did you build the adapter yourself?
> 
> Also, check whether you get the same issue on the RGB video output.

They were right. It was the plug-in lead from the Pi Zero. I checked it with a Pi 3 A+ and it worked great. 

Thanks for your great project!

Greetings

Sascha

closed 2022-12-21 by IanSB