IanSB/RGBtoHDMI #14: Apple IIc colors inverted/wrong with Beta54

jma7x, 2022-11-01:
I just tried Beta54 with my Apple IIc (TTL interface) with a Pi 3 B, and the colors are somehow inverted/wrong.
I see the Artifact Type defaults to "Apple (Zero 2W)".  If I change it to "Simple (Old Code)", then it seems fine.
This affects both text and graphics.  (Note: I also tried Beta53 and it seemed to only have the issue with graphics)

So I guess the question is, should the new artifact color work with the Pi 3 B (from the release notes, it sounds like it?)

[image]

Thanks

IanSB, 2022-11-05:
@jma7x 

It looks like the new artifact code can't handle the inverted video on the IIc's digital output. I don't have a IIc but I can pick up the same signals internally on the IIe so I will have a look at that when I have made up a lead.
The new code should work on the analog video output which is not inverted but would require an analog board.

jma7x, 2022-11-06:
Unfortunately I don't have an analog board, so I can't try that.  I can test any updates you make though for the digital output if that would help.
Would it be feasible to just have the cpld invert the signal?

IanSB, 2022-11-06:
@jma7x 
>Would it be feasible to just have the cpld invert the signal?

No, the fix would have to be be done in software.

IanSB, 2022-11-26:
@jma7x 

FYI I've just made up a lead so there should be a fix soon now that I can test it

IanSB, 2022-11-28:
@jma7x 

There is a new beta 55 with a fix for Apple IIc colors here:
https://github.com/IanSB/RGBtoHDMI/releases

Let me know if it works for you

jma7x, 2022-11-29:
Looks good with beta 55!
Switching back and forth between the old code and the new, solid blocks of color seem the same, but I do see the subtle improvements on edges with the new code.
Thanks!

closed 2022-11-29 by jma7x

IanSB, 2022-11-29:
@jma7x 
>Looks good with beta 55!
Switching back and forth between the old code and the new, solid blocks of color seem the same, but I do see the subtle improvements on edges with the new code.

Thanks for confirming.
The new code is mainly meant for CGA sources as it works properly in 320x200 4 colour modes and also displays the >1000 colours generated by the 8088mph demo. The old code only works in mono mode mode on CGA (i.e. similar to the Apple II) and displays a maximum of 16 colours.