IanSB/RGBtoHDMI #20: PAL Artifacts with c64videoenhancement board

thbreidenbach, 2023-10-14:
Hi, just tried the beta 60 with c0pperdragons videoenhancement board and Analog Board/RGBtoHDMI. All works perfect but PAL Blending seams to be off for this version and trying different settings doesn’t change the output for in different test images that use pl blending.
Is there a way to reproduce PAL Artifacts in this constellation?
Regards 

Thorsten

IanSB, 2023-10-16:
@thbreidenbach 

Are you using c0pperdragon's new lumacode board or the old YUV board?

The PAL artifact only works with the new lumacode board at the moment. I just tried Beta 60 with that and it does show a difference on my system although it is quite subtle (some squares are different):

Artifact Off
[image]

Artifact On:
[image]

thbreidenbach, 2023-10-17:
I use the yuv board because I didn’t know the other. We’re hardware enthusiasts so building was mandatory..
Will you still Support yuv and are there chances for Artifact support?

IanSB, 2023-10-19:
@thbreidenbach 

Try this alpha 61B release which has PAL artifacts for the YUV profile:
https://github.com/IanSB/RGBtoHDMI/issues/21#issuecomment-1769689216

thbreidenbach, 2023-10-20:
That works quite well! I had some trouble finding the right palette settings (Commodore64 instead of C64yuv) and I'm still new to the settings.
Theres still some parts that seem to be off, but it looks very promising!
[image]

IanSB, 2023-10-20:
@thbreidenbach 
>That works quite well! I had some trouble finding the right palette settings (Commodore64 instead of C64yuv) and I'm still new to the settings.

Sounds like you just overwrote the sd card without wiping it first. The names of some palette and profile files have changes so if you don't wipe you will end up with old profiles and palettes mixed in with new ones.

You only need to select the "Commodore 64 YUV" profile and everything should be set up correctly.

The are two settings in the palette menu that affect things:

PAL Odd line: 
This sets the behaviour of the PAL Odd lines which have a different phase to the even lines
Off =same as even
Blended Colours: only different colours are blended
All Colours: all colours are blended which can give hanover bars on solid colours

PAL Odd Level:
This changes the phase offset of the odd lines compared to the even lines.
Setting this to 0 is the same as setting PAL Odd line to off.

thbreidenbach, 2023-10-26:
Thank you!! Thats about right.. Wiped the card, copied, result is far better. I'm still new to the concept and need to fiddle around. Would it be possible to change the special c64yuv palette interpretation from within rgbtohdmi config?

The results are really great! In some demos it seems like there's a flicker effect in High Res Images. The picture on composite shows no flicker. Could be intended, hard to tell.

IanSB, 2023-11-12:
@thbreidenbach 
That feature is now in the new beta 61 release:
https://github.com/IanSB/RGBtoHDMI/releases