IanSB/RGBtoHDMI #22: Artifact Colors with Atari 800 Lumacode ProfIle

scorpio-ny, 2023-11-04:
Hi @IanSB ,

I have enabled artifact colors to use with some games that use color artifacts with the Atari 800 Lumacode profile. I am using the latest beta firmware. When I turn it on artifacting in the palette menu and try to set the colors be switching the angles, nothing seems to happen. Is there something else I need to do?

IanSB, 2023-11-04:
@scorpio-ny 

Artifacting is not yet supported on the Atari 800.
Currently NTSC artifacting is supported on mono (e.g Apple II / Tandy CoCo) and CGA (IBM PC)
PAL artifacting is supported on 16 colour lumacode profiles such as the Commodore 64

What Atari 800 programs use artifacting?

scorpio-ny, 2023-11-05:
> @scorpio-ny
> 
> Artifacting is not yet supported on the Atari 800. Currently NTSC artifacting is supported on mono (e.g Apple II / Tandy CoCo) and CGA (IBM PC) PAL artifacting is supported on 16 colour lumacode profiles such as the Commodore 64
> 
> What Atari 800 programs use artifacting?

Hi @IanSB ,

Quite a few games used them since a lot of them were ports from the Apple II back in the day. Here is a partial list:
Ultima II, Ultima III & Ultima iV
Flight Simulator II
Drelbs
Pinball Contuction Set
Choplifter
LodeRunner
Snakebyte
Wavy Navy
Night Mission Pinball
Hard Hat Mack
A.E.
Threshold
Lunar Leeper
Aztec

IanSB, 2023-11-06:
@scorpio-ny 

Are these games all in mono like an Apple II or do they mix the mono with other colours?

scorpio-ny, 2023-11-07:
> @scorpio-ny
> 
> Are these games all in mono like an Apple II or do they mix the mono with other colours?

A little of both. Most of the examples I linked are mono like the Apple II. The Atari can also split the graphics modes on the screen  so part of the screen is mono and the rest has color. On the mono graphics modes games, the color are all generated by artifacts. What is interesting about the Atari is depending if it had the CTIA or GTIA chip or the model of Atari computer, the artifact colors it creates can be different.

Here is a link to an article on the subject that hopefully provides some useful information:
http://www.atarimania.com/faq-atari-400-800-xl-xe-what-is-artifacting_22.html