IanSB/RGBtoHDMI #24: NTSC Atari 800XL with PAL Antic

smarquette, 2023-11-10:
I have a NTSC Atari 800xl with a PAL Antic and I’ve installed c0pperdragon’s GTIA digitizer for HDMI output. I can make it work with an NTSC Antic and GTIA, but if I switch the Antic to PAL, I can’t find a setting that will make this work.

IanSB, 2023-11-10:
@smarquette 

I assume this means you would be outputting PAL frame rate but with the timing derived from an NTSC crystal?
Does the resulting PAL timing produce colours with the NTSC palette or the PAL palette?
To support such hybrid modes would need a couple of extra profiles in the Atari folder, the settings of which depend on the above answers.

smarquette, 2023-11-10:
Yes with NTSC crystal. Color palette is NTSC and I have a Ultimate 1MB upgrade that can force the NTSC colors. I have a few newer games that need to see PAL Antic. Thanks!

IanSB, 2023-11-10:
@smarquette 

Delete the profiles folder on the SD card and unzip then replace with this one (reboot the Pi):

[Updated_Profiles.zip](https://github.com/IanSB/RGBtoHDMI/files/13323521/Updated_Profiles.zip)

This should support the following hybrid combinations but is untested:

NTSC GTIA with NTSC crystal and PAL Antic
PAL GTIA with PAL crystal and NTSC Antic

Let me know if this works and I'll include it in the next beta due soon.

smarquette, 2023-11-10:
Thanks Ian,
I'll give it a try tomorrow.

smarquette, 2023-11-11:
Doesn't seem to work, I've tried all the Atari 800 lumacode profiles. Here's a picture of the screen.

Thanks

IanSB, 2023-11-11:
@smarquette 
>Here's a picture of the screen.

Images don't seem to be included if you reply by email, I think you have to post on the github thread instead.

Can you also post a copy of the source summary page in the info menu. You can screencap this by pressing the up and down buttons (file on SD card in  /captures)

smarquette, 2023-11-11:
OK Here are the pictures:

[image]
[image]

IanSB, 2023-11-11:
OK it looks like the new profile will only work with the latest alpha versions so try this version:

[Alpha61H-with-new-profiles.zip](https://github.com/IanSB/RGBtoHDMI/files/13325827/Alpha61H-with-new-profiles.zip)

IanSB, 2023-11-11:
@smarquette 
Do a clean reinstall of the above  by wiping the SD card and then re-select the Atari profile

smarquette, 2023-11-11:
Seems to be working. 
Thanks!!!

[image]

IanSB, 2023-11-11:
@smarquette 
>Seems to be working.

Thanks for confirming, those profiles will be in the next official beta (Beta 61) which will be released in the next day or so

A couple of questions:
What is the physical resolution and size of your monitor? (It looks similar to some 10" 1024x768 monitors I have)
In the above screenshots it shows resolution set to "Auto" but in the summary menu it shows Pi resolution set to 720x480 which is not a standard LCD resolution as far as I'm aware.

Could you use the save log.txt and EDID.bin option in the source summary menu and post those two files from your sd card. 

Also do some PAL Atari games use more than 200 active lines (i.e. more active lines than an NTSC equivalent)?

smarquette, 2023-11-11:
The monitor is a 12” HD TV that can do NTSC and PAL. Since the last screenshot, I’ve changed the resolution to 1280x720 and the monitor reports it as 720p. 60hz.
If I connect my PAL C64 to my RetroTink 2x it displays as 576p.

I’ll upload those menu screen captures when I get back tomorrow morning.

As for the PAL games, I have a few, but I’m not sure of number of lines. The only game that wont run on a NTSC Antic is Bubble Shooter. The others just run at a slightly faster speed. The NTSC games all run at a slightly slower speed that is hardly perceptible.

IanSB, 2023-11-12:
@smarquette 
I found this monitor on Ali Express:
[image]
Is that it? It's available in 800x600 and 1024x768 resolutions

If so you should try setting the resolution to the correct one of those two. One will look extremely sharp, the other will have softer edges.
RGBtoHDMI produces the sharpest results when set to the native physical resolution of the monitor but in that integer scaling mode some combinations of computer resolution and monitor resolution will produce larger borders. In this case 800x600 will have larger borders than 1024x768 so the latter will give the better results if your monitor is also 1024x786.

smarquette, 2023-11-12:
That looks just like it. I got it on Amazon a few years ago. I’ll give changing the resolutions a try and see if it improves.

smarquette, 2023-11-12:
Here are the log files you requested.

[logfiles.zip](https://github.com/IanSB/RGBtoHDMI/files/13328148/logfiles.zip)

This version of the monitor is actually a TV and can do up to 1920x1080 (1080p) resolution as well as standard computer resolutions.

I'd like to find a resolution/scaling setting that will fill most of the screen. Right now there is just a box in the middle with a 2-3cm border. I can use the zoom menu in the TV's menu to fill most, but it doesn't retain the setting when I turn it off.

Thanks again.

IanSB, 2023-11-12:
@smarquette 
>Here are the log files you requested.

Thanks, the monitor is detected as 800x600 physical resolution so that should be the resolution selected when set to auto:
```
Timing descriptor detected as 800x600
```
Is that the case?

>This version of the monitor is actually a TV and can do up to 1920x1080 (1080p) resolution as well as standard computer resolutions.

It will probably accept a large range of resolutions and then scale them (interpolate) to fit the 800x600 screen

>I'd like to find a resolution/scaling setting that will fill most of the screen. Right now there is just a box in the middle with a 2-3cm border.

Yes, that is the problem with such a low resolution monitor, because there are so few pixels the computer resolution doesn't fit nicely into the screen resolution for integer scaling (where one computer pixel is mapped to an exact number of LCD pixels) so you end up with larger borders. This is less of a problem with higher resolution screens like 1920x1080.
See the opening page of the wiki for the difference between integer and interpolated scaling:
https://github.com/IanSB/RGBtoHDMI/wiki

To fill the screen you would have to change the scaling setting to Interpolate 4:3/Soft in the main menu and then use the crop border (zoom) setting in the preferences menu to crop off as much of the border as you prefer (use save configuration to save that setting)
When using interpolated scaling you can change the aspect ratio a little by altering  the Max Height and Max Width settings. Again use save configuration to save that setting

You may get better results by changing the scaling setting to Interpolate Full/Soft and trying all the other resolutions including widescreen ones (followed by the crop border (zoom) setting if required) as they may get squashed to fill your 4:3 screen.
 
BTW there is now a new beta 61 official release with the hybrid Antic support:
https://github.com/IanSB/RGBtoHDMI/releases