IanSB/RGBtoHDMI #28: MZ0404 audio embedder

NemoN, 2024-02-06:
@IanSB 
Do you have more information about the MZ0404 audio embedder mentioned in the last release?
Unfortunately I can't find a shop for it...

IanSB, 2024-02-06:
@NemoN 

It's not actually sold under that name but that's what appears in it's EDID info

I got it from here:
https://www.aliexpress.com/item/1005005604559797.html
It does appear to be listed elsewhere like ebay etc.

IanSB, 2024-02-06:
@NemoN 

Just to add that you should manually specify the resolution and refresh rate when using it, not rely on the Auto setting which might not always work
i.e. 
Set resolution to 1920x1080 & Set Refresh to 50Hz when using a HD monitor with a 50Hz computer etc.