IanSB/RGBtoHDMI #30: Unable to see new profiles on the profile menu.

Finklewart, 2024-03-22:
Hi, 
I've created a custom profile and can use it fine loaded as custom profile 1.
I followed the instructions on the wiki to move it to the normal profile folders and renamed it to a sensible name but whatever i do i can't get it to show up on the normal profile menu.
I'm sure I'm doing something silly.
I've got the correct 6-12_BIT_YUV_Analog folder and ive not renamed the extension from .txt
I notice a lot of the folder and file names are using _ instead of space but i tried folder and file names without spaces and that didn't work either.

Many thanks,

Jim

IanSB, 2024-03-23:
@Finklewart 
What RGBtoHDMI board are you using? (i.e. main CPLD + analog board or the special lumacode/mono board)
If it's the lumacode/mono board then you need to add a trailing _ to the filename before .txt to make it show up in the profile lists. plus any new machine name folder (like Commodore / Apple etc) also has to have a trailing _

So if your machine name was Foo and the profile was Bar.txt then you need to create a folder:
/Profiles/6-12_BIT_YUV_Analog/Foo_/Bar_.txt

The reason is that both the analog board and the lumacode board share the same profile folder but only a subset of those profiles will work on the luma board so to avoid confusion profiles that don't work (i.e. without the trailing_ ) are hidden.

Leading _ is also used to change the sorting of files and folders so that special folders like Test and Custom appear at the end of the list and not in between normal manufacturers.

All the _ characters are removed or replaced by spaces in the menus so this is normally hidden from the user.

BTW what is your custom profile for?
If it's an unsupported machine perhaps you would post it here for inclusion in a future release.

Finklewart, 2024-03-23:
Hi, thank you for your reply.
It's just using the analog board, the same one as used on the ZX80/81/Apple/UK101/Ohio etc.

So for the normal analog board i don't need to worry about the trailing _ ?

The machine is a 911VDT card in a Texas Instruments TI-990 system, it has a composite video output.
Once I can get it to appear in the menu correctly i will post it.

IanSB, 2024-03-23:
@Finklewart 
If it's the normal analog+cpld board pair with the 6 way IDC connector then it shouldn't need the trailing _

What is the exact path + filename you are using?

Finklewart, 2024-03-23:
Hi, 
The analog board i have has a single 3.5" connector.
Is that the lumacode board ?, the person i bought it from didn't mention lumacode.
Ah, i've added the trailing _ to the path and the file name and it now shows up in the profile menu.
I had tried the _ on the file name and the folder name but i don't think on both at the same time.
Thank you very much.
I didn't see the mention of the _ in the tutorial for adding new profiles.

The file is attached, this is for the VDT911 board jumpered to 1920 characters (80x24) and 50 Hz, i will add the other 3 modes at a later date, 906 character (80 x 12 !) and both in 60 hz

Thanks for the help, 

Jim

[Texas_Instruments_TI-990_VDT911_card_.zip](https://github.com/IanSB/RGBtoHDMI/files/14733354/Texas_Instruments_TI-990_VDT911_card_.zip)

IanSB, 2024-03-23:
@Finklewart 
>The analog board i have has a single 3.5" connector.
>Is that the lumacode board ?

Yes. It's printed on the PCB and shown in the System Summary menu:
[image]

>The file is attached, this is for the VDT911 board

Thanks, it will probably appear in the "Other" folder for now.

>I will add the other 3 modes at a later date, 960 character (80 x 12 !) and both in 60 hz

It is best to create each profile individually but if the timing of these profiles is sufficiently different you can put them all in a sub-folder and RGBtoHDMI will auto select the correct one. This will certainly be the case for the 50/60 Hz ones but I don't know what difference there will be between the 80x24 and 80x12 ones. (There might not be any difference in which case both modes will work with the same profile)

See existing profiles for an example of the structure of this e.g:
/Profiles/6-12_BIT_YUV_Analog/Sinclair_/ZX81_Composite_

>I didn't see the mention of the _ in the tutorial for adding new profiles.

I haven't got around to documenting it so far. Actually, looking at the current profiles there are now a lot more uses of the _ than 'normal' profiles so it may be better to reverse the logic of this so the _ is used to signify the non-mono/luma profiles.
I might change that in a future release.

Finklewart, 2024-03-25:
Sorry for causing confusion, i'd not realised i was using the the lumacode version of the board.
Feel free to change the file and pathnames.
I think you are right, the 80 x 24 and 80 x 12 use the same screen resolution just with a lot more padding on the 80 x 12 characters.

Thanks agaon,

Jim