IanSB/RGBtoHDMI #32: P6 on Lumacode Board

venice1200, 2024-06-18:
This is not an issue!
I like to to known what function has P6 on the Lumacode Board.

[image]

Many Thanks

IanSB, 2024-06-19:
@venice1200 

It's connected to the reset button for use with the hardware reset option on models other than the Pi zero.

For the Pi zero you can solder a 2 pin header on the zero board and a matching 2 pin socket in the two holes to the right of P6  (which is actually P4 on the underside) and that will allow the hardware reset option to work.

Other Pi models like the zero2W or Pi 1 to Pi 4 don't have the matching 2 pin header on the Pi board so you have to use a cable to link up from P6 to the reset input on those Pi boards (the location varies from model to model).

The lumacode board and the latest issue 7 normal CPLD board support software reset when the reset button is pressed so making the hardware reset mod is no longer essential and it's really only useful during software development to recover from crashed software when the software reset won't work.

venice1200, 2024-06-19:
Many Thanks for the information!

And all the best

closed 2024-06-19 by venice1200