IanSB/RGBtoHDMI #35: Atari STe - can a 12 bit extender support Mono and Colour?

shooking, 2024-11-05:
Hi there

fundamental question - should the RGBtoHDMI with 12 bit extender, suitably wired to an STe, allow low/medium colour and mono (shorting monitor 4 and 13)?

For my STFM I got it to work - had to modify profile on one system, worked with another "out of the box"

For my STe - colour worked when I chose the STe profile.
Mono - I get just a black screen, verified via screen capture, and grid lines with debug. But I see the analog out.
If I change sampling to 9 bits vsync I see signal partially getting thru
 
[image]

Hints appreciated. I am about to wire in just the 0v 5v, mono and csync as a test. BUT there seems to be a lot of debate regarding the RGB resistor ordering. See my notes at end.

Finally - would be great if WIKI showed pinout of the 16 pin male connector for the 12 bit extender.

MY SYSTEMS

So I have two types of RGBTOHDMI systems

1. RGBtoHD 12 bit issue 4 2018 - 20 Hoglet & IanB
2. RGBtoHD 12 bit issue 7 2018 - 23 Hoglet & IanB

And two 12 bit extenders

1. 12 bit adapter for RGBtoHD Issue 4 2022 IanB - struggled with STFM
2. 12 bit adapter for RGBtoHD Issue 2 2020 IanB - worked with STFM, colour STe but not mono

1. Atari STFM modified to 4Mb, CO78789
2. Atari STe CA 4003290 1989

and final part of "complication" I have

1. Pi Zero W V1.1 2017
2. Pi Zero 2 2021

WHAT WORKS

Atari ST worked with with a lot of help from Mikerochip and Javi.

I used shifter chip.
Issue 2 2020 adapter just worked.
Issue 4 2022 worked with colour. Needed help with mono
[reduce sample rate to 3bpp, modify palette to get b/w rather than yellow]
Pi choice was irrelevant (other than 1.1 allows reset)

I moved to the STe - and well things are a bit odd.

I can get colour but not mono.
Just a black screen.
Capture shows black.
Debug, show scanlines, show genlock ... displays something.
I feed the analog signal to a VGA monitor and the picture is clear.
The RGB is black BUT I modified AtariSTe Sampling to 9Bits Vsync and you can hopefully see the mono image in background. albeit geometry and position is off.

Any hints gratefully appreciated.

ONE HINT - would be nice to have a diagram of the IDC pins.
I believe they are "standard" but there is some ambiguity on the silkscreen. There is standard triangle with corresponding pin 1. But then 4 pins along there is P4. It _could_ mean ordering is Pin1 2 3 4 along long row. But seems to be normal pin 2 BELOW pin 1.
A diagram on the wiki would be cool.

I have researched and found official cable agrees with Mikerochip's hints

[image]

But I see other suggestions from pixelk - perhaps using an Amiga board?

[image]

Here's my comparison of these suggested resistors

[image]

And here's my photo (sorry I am red/green colour blind so I make it same as above chart BUT I am confused why there is a blu2, B2 then recommendations to change ordering due to STe R3 really being R0?
Could it be issue 4 fixes problem with issue 2 boards?

[image]

shooking, 2024-11-05:
I saw that thanks. I wasn't sure if the 12 bit extenders I got would work. If not (on the STe) I am OK with that - I will use MONO via VGA analog adaptor. All my monitors can handle it. I only have one monitor that could handle colour analog.

On your excellent write up I didn't understand the need to wire directly to chip. I will.re read.

I also don't understand why STFM works, but STe doesn't. I wonder if it is because the STE is using CSYNC rather than individual h and v.

luciust, 2024-11-05:
Actually - let me know which part was *not clear* - happy to change it if helps - primarly those are notes for me... So I don't have to remember it all. 

CSYNC, or composite sync, can be described as a combined version of the HSYNC and VSYNC signals. Since CSYNC is defined by those two sub-components, and as you know HSYNC is horizontal sync and VSYNC is vertical sync and it is combined later for the RGBtoHDMI into CSYNC anyway. So - maybe you need to take a look at the STE profile? Show me your STE picture output and settings, maybe we will figure it out that way.

Definetly using the Amiga pickup solves any STE issues. Go for that! PCBs are very cheap today. I would send you one but it still would be more expensive than any 5 pcs from China.

shooking, 2024-11-05:
Hi and thanks.
What I am unclear about is if the 12 bit extender board I got can be modified to allow mono. My guess is the STFM version, using the level shifter and separate syncs, is doing something different to the STe.

I also see some discussion about R0 to 3 being labelled incorrectly on an STe.

I understand syncs. From what I read the STe profile knows a CSYNC will be passed in. This leaves the spare pin to interpret mono. I was hoping that the 12 bit extender I got would work in mono mode.

I will see if I can hack my setup and/or order Amiga PCB.

Thanks again

luciust, 2024-11-05:
I think it is correctly passing the MONO signal to the RGBtoHDMI sampler, but my mind is elswere now. It is hooked up to bus transciever and then... sorry - magic happens. I think the actual answer might be in my communcation couple of years back on this or on hoglet76 github, so if you can find my old posts - that would probably shed some light.

IanSB, 2024-11-05:
@shooking 
>fundamental question - should the RGBtoHDMI with 12 bit extender, suitably wired to an STe, allow low/medium colour and mono (shorting monitor 4 and 13)?

Yes it should work with the mono signal fed to the Vsync input of RGBtoHDMI which is repurposed as a video input for mono mode. The mono sub profile should be auto selected when the sync signals switch to the higher refresh rate so confirm that the sub-profile shows as the mono one. (Auto selection is disabled when the menu is on screen so only call up the menu after switching)

>Hints appreciated. I am about to wire in just the 0v 5v, mono and csync as a test. BUT there seems to be a lot of debate regarding the RGB resistor ordering. See my notes at end.

The problem with bit ordering is due to the way that the STE schematic numbered the bits.

On RGBtoHDMI the bits were always R3, R2, R1, R0 ; G3, G2, G1, G0 ; B3, B2, B1, B0 with the '3' being the most significant bit  and '0' being the least significant bit.

On the ST there were only 3 bits each of RGB and they were numbered R2, R1, R0 and they could be mapped easily to the existing RGB inputs.

However on the STE an extra bit for each of RGB was added and this was the least significant bit but it was named R3, G3, B3 so the bit ordering was messed up. I presume this was so that the most significant 3 bits matched the ST but it meant that the bit order was R2, R1, R0, R3 etc.

The confusion arose from that as some users mapped them to RGBtoHDMI directly by number which messed up the bit order.
Whereas the correct order is RGBtoHDMI R3, R2, R1, R0 to STE R2, R1, R0, R3 etc.

The easiest way to avoid confusion is to go by resistor values:

The smallest value RGB resistors are connected to RGBtoHDMI R3, G3, B3
The next largest value RGB resistors are connected to RGBtoHDMI R2, G2, B2
The next largest value RGB resistors are connected to RGBtoHDMI R1, G1, B1
The largest RGB resistor values are connected to RGBtoHDMI R0, G0, B0

shooking, 2024-11-05:
Thanks @IanSB 
I put diagrams based on your cable pins.
I traced the 16 pin IDC to the R0 G0 B0 etc that flow to the 12 pin header and 4 pin riser.

Did u see my partial mono image if I chose 9 bit csync?

I am wondering if the mono is generated by the low RGB. I saw your reply to @bikerbob (same as you suggested above). I originally wired according to diagram above. But I can see from the resistors values this is incorrect when compared to your explanation. It worked - as in  I got acceptable colour but I am red green colour blind :-(

Tomorrow I will solder some 90 deg header to the resistors ends and experiment with dupont cables.

Device works flawless on the STFM. Just can only get poor mono using 9 bit and vsync. Analog his res is fine. So I will wire according to resistor colours as per hints.

Thanks again for support.
Would be great to have the IDC pinout diagram  added to the wiki.

Best regards
Steve

IanSB, 2024-11-06:
@shooking 
>Did u see my partial mono image if I chose 9 bit csync?

In that mode, the vsync input is used as one of the Green bits so that likely means that you have connected the mono signal to the vsync input correctly so maybe the problem is that the mono profile (which makes RGBtoHDMI look at the vsync input for video) isn't being selected properly.
Try setting Autoswich to "Off" and then manually change the sub-profile to the mono sub-profile (when the STE is in mono mode)

Also post a screencap of the source summary page in mono mode and use the info menu to save the log and EDID and post log.txt from the SD card.

shooking, 2024-11-06:
OK so we remapped the R0 - B3 based on resistor values as per your suggestion. The colours seems nicer - I got a non colour blind friend to check.

Whilst doing this I first wired in just the 0v 5v sync and mono then used analog cable to force hi res.

Same behaviour - the mono only partially shows if STe Mono 50Hz profile is chosen and that seems to set sampling to 9 bit with sync.

Save configuration ignores these settings and I need to apply them each time.

The mono image would need H and V adjustment. Again - this seems to be ignored when I exit that menu - I will read.manual to see what I am doing wrong.

I am using latest release (February) from the main hoglet github.
Is there a specific CPLD version is shod be using? 93 (9.3) seems to be the one.

Once I know I am.using the right firmware I will save and provide the log info.

My STe is a 50 Hz PAL version. I will boot up diag via Sidecart tomorrow and see if the hardware is set to 50 or 60 Hz.

Thanks again for your support.
That RGB mapping is tricky!
The STFM was much more straightforward.

IanSB, 2024-11-06:
@shooking 
>OK so we remapped the R0 - B3 based on resistor values as per your suggestion. The colours seems nicer - I got a non colour blind friend to check.

You can check this properly using a diagnostic cartridge or program which displays colour gradients like this:
[image]

>Is there a specific CPLD version is shod be using? 93 (9.3) seems to be the one.

RGB 9.4 is the current version (can be manually reprogrammed via the settings menu)

>That RGB mapping is tricky!

I have updated the cables section of the wiki to note all the above

shooking, 2024-11-07:
Hi @IanSB 
Thanks for help and updating wiki.
Here's color bar from STe via STe diag cartridge.

[image]

I also screen compared Hatari with STe/RGBTOHDMI and looks good.

Regarding the mono.
I see nothing when mono is selected via RGBTOHDMI but I see the analogue output. The only time I see output is if I select 50Hz subprofile. The sampling seems to change to 9 bit vsync. I read  your wiki notes yesterday. I don't seem able to find the 1bit vsync (or anything less than 3 bits)

I will send logs once I extracted them.
But colour modes are looking real good. Will solde that fix in over the weekend.

luciust, 2024-11-07:
For mono mode STE (or ST) are the same: 640×400, H @ 35KHz, and V @ 71Hz - so 71Hz is the expected value.

luciust, 2024-11-07:
> Hi and thanks. What I am unclear about is if the 12 bit extender board I got can be modified to allow mono. My guess is the STFM version, using the level shifter and separate syncs, is doing something different to the STe.

OK, so this part migh be unclear, also pardon if I'm repeating the steps you've already done.

First With STE you need to connect the MONO signal directly to the IC's leg. Since this is Amiga pickup board it was not ready 100% for STE, but if you get the number 4 resitor signal to a unused via (4/L) and then solder the wire directly to te "L".

[image]

If mono mode is used on ST/STE - the other (RGB) signals are not used, either for monitor or RGBtoHDMI - the signal is just the mono.

Finally - you need to tell ST that it should *start* in mono mode - meaning that the monitor plug needs to have either wiring or have the mono monitor cable connected, for reference check out:
- how mono is detected by Atari ST 
- where mono signal is present:

[image]

shooking, 2024-11-07:
thanks @luciust 
At the moment I only have the IanB's 12 bit extender. Is the chip the same as one used on the Amiga board? If so - maybe I can try to touch a wire to mono on that pin 4.

I ran my STFM onto same monitor using same SD Card, verified settings and saved files.
Then I did same on STe - I think the log files are appended?

I then tried copying the ST's mono profile to the STe - I had to change the sync to composite but still no joy,

The only way I can get a partial picture in mono is using the 50Hz sub profile.
I will attach the files to see if anyone can suggest an edit that might work.
Else - I am happy with the RGB and will use the STe in colour mode HDMI and mono mode VGA and hope someone else finds a solution.

[MyRGB.zip](https://github.com/user-attachments/files/17667517/MyRGB.zip)

zip should contain EDID.bin, log, profiles including the ST as working with STFM and mono.

If tried copying that mono to the STe - didnt appear to work. I also modified the sync to composite and that didnt work.

So far the 50Hz STe is the only one that gets me a picture.
Yes the refresh is 71.2-71.4Hz vertical.

IanSB, 2024-11-08:
@shooking 
>Yes the refresh is 71.2-71.4Hz vertical.
>zip should contain EDID.bin, log, profiles including the ST as working with STFM and mono

I can't actually see it working at 71Hz in the logs, only at 50Hz but it can get confusing with the auto switching of profiles so can you try the following:

Unzip the attached file and put the two profiles in \Profiles\6-12_BIT_RGB\Atari on the SD card and reboot the Pi.
This will add 2 separate profiles in the Atari section of the profiles menu:
TEST_Atari_STE_50Hz
TEST_Atari_STE_Mono
(not to be confused with the sub-profiles of the existing STE profile)

[TEST_Atari_STE.zip](https://github.com/user-attachments/files/17671317/TEST_Atari_STE.zip)

Then do the following in this order:

First switch your Atari STE into colour mode
Go into the Atari section of the Select Profile menu and select TEST_Atari_STE_50Hz profile
Go into the source summary page in the info menu and screencap the info (Press the up & down buttons together)

Then switch your Atari STE into mono mode
Go into the Atari section of the Select Profile menu and select TEST_Atari_STE_Mono profile
Go into the source summary page in the info menu and screencap the info (Press the up & down buttons together)

The screencaps will be in /captures on the SD card in the two TEST... folders

That will give me the exact timing info to confirm that it is set correctly in the profiles.
Once I know the timing is correct I can look further at the blank screen problem.

IanSB, 2024-11-08:
@shooking 

Also I just noticed that you are running the latest software but your CPLD is out of date and should be 9.4
```
Running GPU capture build. Kernel version: 20240213 e630d36f
CPLD  Design: 6-12_BIT_RGB
CPLD Version: 9.2
```

Running an old CPLD with new software is not a tested combination and I think mono video on vsync is not supported in that old CPLD version so try this first:
Go into the Settings menu, then select the Update CPLD menu and program RGB v94

Do this before the tests above.
I would still be interested in the screencaps even if it now works so I can check the timings.

shooking, 2024-11-08:
@IanSB and @luciust 

How can I thank you both?!!

I took a fresh 64GB card.
Formatted to fat 32
applied software - updated - I see I may have been not reading the YUV part properly
Applied 94
Copied your profiles to same directory as the official profiles
Colour - works
Mono - works!!

Here's a zip of the captures - I grabbed the log/EDID too

[TESTS.zip](https://github.com/user-attachments/files/17676156/TESTS.zip)

So what was my most likely mistake?
Having the wrong CPLD?
And thanks for everyone's help with that funky RGB.

I would ask for one more addition to the wiki please.
Something that shows the generic IDC numbering - in this case I have added the specifics for the Atari STe - but the Pin 1 with arrow (standard) coupled with the P4 (silkscreen) in 4th position made me think that perhaps the pins are numbered sequentially along long rows!

[image]

Anyhow - I can now do the soldering and screw the STe back together.
Great support everyone.

shooking, 2024-11-08:
We also reprogrammed all of my RGBTOHDMI to 9.4 and all works.
I think the 9.4 came out in Feb and I bought my system before then - hence why I had the mismatched CPLD.
The other system also didnt work with the STe until we updated to 9.4 - perfect

Finally - here's the colour bars you will get IF one uses the schematic R0, R1, R2, R3. My non colour blind mate helped me - but even I can see the gaps :-)

[image]

So all solved

luciust, 2024-11-08:
It's all about @IanSB way of doing things - he's has a great way of helping here with perfect results, I could say the same thing when I first looked at the project: how can I thank him! So, inspired by his work I took extra step to document my strugle.
And I'm so glad that another STE is fitted with RGBtoHDMI, as the new racer game is comming!

IanSB, 2024-11-08:
@shooking 
>Applied 94
>So what was my most likely mistake?
>Having the wrong CPLD?

The mono issue was just the wrong CPLD. The software should auto prompt to update the CPLD on a fresh install but there was a bug introduced some time ago that stopped that from working. That has now been fixed for the next release.

>Here's a zip of the captures - I grabbed the log/EDID too

Those timings look fine so I guess the standard Atari STE profile should work OK

>Anyhow - I can now do the soldering and screw the STe back together.

Glad you got it sorted out.

shooking, 2024-11-10:
closing as YES Atari STe works with 12 bit extender provided correct CPLD and wiring is used.
Excellent work all involved.

closed 2024-11-10 by shooking

IanSB, 2024-11-10:
@shooking 
FYI I updated the cables sections of the wiki showing some photos of my Atari STFM installation including use of a detachable cable