IanSB/RGBtoHDMI #36: Amiga A2024 format?

reinauer, 2024-12-20:
Hi, 

is it feasible to implement the A2024 monitor support on RGB2HDMI?

IanSB, 2024-12-21:
@reinauer 
It might be possible but I don't have any hardware to do any tests with.
Do you have any tech info on the resolution, refresh rate, sync type, video levels/bits per pixel etc?

MrSVCD, 2025-04-18:
I would also like A2024 mode.
Here is some resources that I found:
The patent that it is based on: https://patentimages.storage.googleapis.com/48/cd/38/ecc974a1db5c37/US4851826.pdf
Hedley Davis (Designer of A2024) on the A2024:s greyscale: https://forum.amiga.org/index.php?topic=73629.0
A add-in card for A2000,A3000 and A4000 that does the same thing as A2024 but with a different monitor: https://amiga.resource.cx/exp/moniterm

MrSVCD, 2025-04-18:
I did some test in a AmiBerry and recorded a A500 OCS in A2024 mode and notised that the settings for which mode is used and which part of the screen is uppdated is transferred in the top overscan bit of the video signal.
[10Hz-mode.zip](https://github.com/user-attachments/files/19816738/10Hz-mode.zip)
[15Hz-mode.zip](https://github.com/user-attachments/files/19816739/15Hz-mode.zip)

IanSB, 2025-04-19:
@MrSVCD 
>I recorded a A500 OCS in A2024 mode

How do you put an A500 OCS into A2024 mode?

It seems it should be possible to get this working, the only potential problem I can see is that the higher memory bandwidth required for the Hi res screen might interfere with video capture.

I might look at this once I have finished HDMI audio support.

MrSVCD, 2025-04-19:
What you see in the pictures are the test image from AmigaOS 3.2 and in that and I think from version 2.x you can just move the A2024 monitor definition files from storage in to devs.
On a Kickstart 1.3 machine you need a "A2024 Monitor Setup 'Jumpstart' Disk": [A2024 Monitor Setup 'Jumpstart' Disk.zip](https://github.com/user-attachments/files/19821733/A2024.Monitor.Setup.Jumpstart.Disk.zip)
Since I don't have a A2024 to test with I am not sure that it works but in Amiberry it does flicker almost the same way that AmigaOS 3.2 does.

MrSVCD, 2025-04-19:
> It seems it should be possible to get this working, the only potential problem I can see is that the higher memory bandwidth required for the Hi res screen might interfere with video capture.

I think it does use the same bandwidth as the normal signal since it is buffering the each fourth/sixth of the screen and just updates the part that is sent at that moment.
In NTSC land:
The 10Hz mode splits the screen into 6 parts and updates each part 10 times a second, 6 parts * 10 times a second = 60Hz.
15Hz mode does the same but 4 parts and 15 times a second.
(If you try to watch video it would be screen tearing galore.)

In PAL land it is 50Hz divided by 6 (8⅓Hz) and 4 (12½Hz) instead.

MrSVCD, 2025-04-19:
[WB1.3 Jumpstart.zip](https://github.com/user-attachments/files/19822785/WB1.3.Jumpstart.zip) - Amiberry output of the Jumpstart floppy on KS1.3 rom.

Notes on the resolutions:
On OCS hardware with Angus the horizontal resolution is 1008 pixels, On hardware that has Fat Angus or newer the horizontal resolution is 1024 pixels.

On a PAL machine the vertical resolution is 1024 lines and on a NTSC machine it is 800 lines. 
This corresponds to the maximum vertical resolution of the interlaced standard modes *2.

IanSB, 2025-04-20:
@MrSVCD 

Looking at your screencaps of the individual screens, they both seem to be displaying 1008x1024 so why are there two different ways of sending the same resolution with 4 frames and 6 frames using (I presume) 504x256 and 336x256 respectively?

Also do you know how the Amiga's 12 bit RRRRGGGGBBBB digital output which is captured by RGBtoHDMI maps to the RGBI digital output of the video hybrid?

MrSVCD, 2025-04-20:
> Looking at your screencaps of the individual screens, they both seem to be displaying 1008x1024 so why are there two different ways of sending the same resolution with 4 frames and 6 frames using (I presume) 504x256 and 336x256 respectively?

The 6 frame mode uses less bandwidth than the 4 frame mode.
This according to the manual [https://www.retro-commodore.eu/monitors/](https://www.retro-commodore.eu/monitors/) "A2024 High Resolution Monitor User's Guide" 
Just the English part: [Commodore_A2024_High_Resolution_Monitor_Users_Guide-English.zip](https://github.com/user-attachments/files/19825019/Commodore_A2024_High_Resolution_Monitor_Users_Guide-English.zip)

> Also do you know how the Amiga's 12 bit RRRRGGGGBBBB digital output which is captured by RGBtoHDMI maps to the RGBI digital output of the video hybrid?

The signal for RGBI is taken (almost) direct from Denise so the adapter-boards have access to them already.
```
DR = R(3)
DG = G(3)
DB = B(3)
DI = B(0)
```
This is according to the schematics available here: [https://amigawiki.org/doku.php?id=en:service:schematics](https://amigawiki.org/doku.php?id=en:service:schematics)
It is a bit hard to see on the A500 schematic but there is a dot on the lines used by RGBI right next to the Video Hybrid to show that the lines on the right is connected under the HY1 IC to the DB23.

IanSB, 2025-04-20:
@MrSVCD 
>Notes on the resolutions:
On OCS hardware with Angus the horizontal resolution is 1008 pixels, On hardware that has Fat Angus or newer the horizontal resolution is 1024 pixels.

In 1008 pixel mode the horizontal resolutions are 504 for 4 frame mode and 336 for 6 frame mode
How does the horizontal resolution of 1024 work?
In 4 frame mode it would be 512 pixels per frame but what about 6 frame mode as 1024 is not divisible by 3?

MrSVCD, 2025-04-22:
More pictures from  Amiberry with a test pattern for width:
[10Hz Fat test.zip](https://github.com/user-attachments/files/19853641/10Hz.Fat.test.zip)
[10Hz OCS test.zip](https://github.com/user-attachments/files/19853644/10Hz.OCS.test.zip)
[15Hz Fat test.zip](https://github.com/user-attachments/files/19853643/15Hz.Fat.test.zip)
[15Hz OCS test.zip](https://github.com/user-attachments/files/19853642/15Hz.OCS.test.zip)
(The OCS version had some graphics glitches for some reason.)

It seems that 6 frame mode is 352 pixels wide that overlaps with 16 pixels, so if you put the pictures side by side without overlap it is 1024+32=1056 pixels.
This seems to be true for both OCS and Fat Angus, just that for some reason the OCS version repeats the first 16 pixels from the previous line in its last 16 so the monitor just cuts them off.

The 4 frame mode is always 512 pixels wide and OCS Angus again has the same bug here.

MrSVCD, 2025-04-22:
Here are the first couple of lines that a PAL Amiga 500 uses to communicate with a A2024:
Fat Angus:
  10Hz 1-bit
[image]
  10Hz 2-bits
[image]
  15Hz 1-bit
[image]
  15Hz 2-bits
[image]
OCS Angus:
  10Hz 1-bit
[image]
  10Hz 2-bits
[image]
  15Hz 1-bit
[image]
  15Hz 2-bits
[image]

The first lines are red and I think that is a sign that extended resolution/A2024 mode signal is sent.
The line with different colours is where different settings are sent:

1. The longer part in the beginning of the line is to set 10Hz (Red) or 15Hz (Black)
2. First short field is which frame that is updated: 0=Black 1=Red 2=Green 3=Yellow 4=Blue 5=Purple
3. The second short field is always yellow on a PAL Amiga 500
4. The third short field is bit depth and if it should crop the last 16 pixels horizontally:
- Red indicates bit depth: Off is 1-bit and On is 2-bits
- Green is always On on a PAL Amiga 500
- Blue indicates if cropping is needed: Off disables cropping and On enables it.

**Note:** I can not visually see if the Intensity bit is set since it is the least significant bit of blue.