IanSB/RGBtoHDMI #38: Beta65 NES graphic errors

mvbhz, 2025-02-23:
I downloaded Beta 65 (Beta65.zip Dec 30, 2024) release to use it on NES (PPUDigitizer & RGB2HDMI). Unfortunately I get graphical errors:

https://ibb.co/356dM3pf

Also the aspect ratio is not preserved - the image is stretched horizontally.

I select the "NES Lumacode" profile. How to fix this?)

IanSB, 2025-02-23:
@mvbhz 
What type and issue number of RGBtoHDMI are you using
e.g. standard CPLD + analog RGB or mono/lumcode board and issue 3, 4 or 5 etc
Are you using the standard or unterminated (unterm) version of the profile. Try the latter if not using it.

mvbhz, 2025-02-23:
I use PPUDigitizer by C0pperdragon + RGBtoHDMI Mono & LumaCode + RBPi0. After switching to RGB profile (Unterm) there are no graphic errors, but there is still aspect ratio 16:9 instead of 4:3.

IanSB, 2025-02-23:
@mvbhz 
What resolution monitor are you using and what are the settings for Resolution & Scaling in the main menu?

Can you post a screencap of the source summary menu (in the info menu)
Press the up and down buttons when in that menu and the screencap will be in /captures on the SD card
Also use the save log and edid option in the info menu and post log.txt from the SD card