IanSB/RGBtoHDMI #40: Beta65 cycling

otacon6530, 2025-03-08:
I am using PPUDigitizer by C0pperdragon + RGBtoHDMI Mono & LumaCode + RBPI400.

It is loading to the color test with 4 squares then all lights go out on the RGBtoHDMI and the screen goes black for a few seconds and then the behavior repeats forever.

Using the latest official release works just fine, except that the nes PPUdigitizer profile doesn't exist of course. Though I do enjoy coco vision via the PPUDigitizer. It has a pleasant if not authentic feel.

IanSB, 2025-03-24:
@otacon6530 

When it goes into a boot loop like that it usually means it can't detect the HDMI monitor but it will reboot once or twice if it needs to change some configuration and if that isn't working for some reason it might continually boot. 

Try a clean install on a new SD card.

To find out what is going on you could connect a TTL USB serial cable to the TxD and GnD pins marked on the 40 way header then view the boot log using a serial terminal program 115200 baud 8 data no parity.

otacon6530, 2025-04-14:
@IanSB The 400 has two HDMI outputs, so I wonder If I just need to switch to the other one or disable the second somehow, so the software doesn't get confused.. I moved the SD card to a raspberry pi 3 for now and it works, so I am less worried about getting the clunky 400 working at this point.

IanSB, 2025-05-20:
@otacon6530 

It looks like the current betas are not working on the Pi 4 due to hardware differences so it may take some time to investigate that.