IanSB/RGBtoHDMI #42: Beta66 - audio add-on board

solarmon, 2025-05-17:
Hi,

Thank you so much for the audio injection feature in RGBtoHDMI!

Not sure if this is the right channel to ask this question, but are there any plans to release the audio add-on board schematics and/or board design file?

Or a way to reduce the price of the boards for bulk purchase?

As a designer and builder of the RGBtoHDMI CDTV v2 beards, I get asked/requested a lot about audio injection into HDMI. It would be great if I can optionally supply these boards with audio injection integration already done so that users/buyers don't need to do it themselves.

My RGBtoHDMI CDTV v2 boards have the audio signals already presented, so integration is certainly easier:

[image]

Thank you!

IanSB, 2025-05-19:
@solarmon 
>are there any plans to release the audio add-on board schematics and/or board design file?

I am looking at at the possibility of releasing an easier to build design for DIY assembly using larger packages at some point in the future but it would be under a no commercial use licence.

At the moment I'm trying to raise some funds to continue this project as it is very expensive to maintain. (I just spent another £1000 over the past few weeks buying additional vintage computers and associated parts just to add support for them). The sellers listed in the wiki provide donations from their sales including the audio board but there are many others sellers that don't provide any financial support. (I haven't had any financial support from any Amiga sellers)

>Or a way to reduce the price of the boards for bulk purchase?

That should be possible, do you have a contact email address?

solarmon, 2025-05-19:
@IanSB 

Please use the email in my user profile.

Thank you for your magnificent work on this project. I have in the past given a donation to c0pperdragon from sales of my boards based on their original RGBtoHDMI board for the Amiga. I'd be happy to contribute for your efforts too.