IanSB/RGBtoHDMI #48: Audio output not detected on Extron IN1608 xi when tring HDMI audio release(s)

AWalGH, 2025-06-20:
Pretty much as is says on the tin. I can confirm that the audio capture boards I've soldered on myself as well ones I have received recently as that were purchased as part of a set all appear to be working just fine for mono audio (It's all I configured them for), but I cannot for the life of me get any of these devices to output HDMI audio to my Extron IN1608 xi  "Presentation Switcher" I use to select and split HDMI between devices (One In From a Choice of Five+Three, Two+One Out).

If I connect the devices directly to one of a few monitors/TVs I have available the RGBtoHDMI devices with one of the newer firmware options all can send audio over HDMI without issue. I also have an ezcap401 that appears to work with the audio output if I connect it directly.

Alpha65G, Beta66 and Alpha67A all appear to produce the same results.

The 1608 has a sub-menu for tweaking audio options since it can mix microphones and other sources, but the menu reports to me that it doesn't detect any audio from the HDMI source when I select one with an RGBtoHDMI.

IanSB, 2025-06-20:
My guess is that the auto detection is getting messed up when connected to the Extron.
Can you use the save log and edid option in the info menu (when connected to the problem switcher) and post Log.txt and EDID.bin from the SD card.

AWalGH, 2025-06-21:
> Can you use the save log and edid option in the info menu (when connected to the problem switcher) and post Log.txt and EDID.bin from the SD card.

Sure thing

[Log.txt](https://github.com/user-attachments/files/20845260/Log.txt)

[EDID.BIN (PixelDrain)](https://pixeldrain.com/u/pXWcHhhB)

IanSB, 2025-06-21:
There doesn't seem to be anything wrong in the logs and I programmed your EDID into an EDID emulator which is connected inbetween a Pi zero and a monitor so that the zero used that EDID info to setup everything and the audio on the monitor still worked so it doesn't look like anything in the EDID is causing the problem.

The only remaining thing I can think of that might cause a problem is an issue with the IEC958 bitstream formatting. Each audio sample is 24 bits and the remaining 8 bits are used to pass some metadata like packet start, sample rate, number of channels, parity bit etc. A lot of TVs seem to ignore this as they work even if all that is missing but some do require it to be present in some form.
Maybe the switcher is very strict about this metadata and if something is slightly wrong then it just ignores the audio. I guess it must be using that data in order to manipulate the audio like mixing in other sources.

Is there any difference when playing the test WAVs or test tone as that uses different code to generate that metadata?

AWalGH, 2025-06-22:
> Is there any difference when playing the test WAVs or test tone as that uses different code to generate that metadata?

Unfortunately no; It still detects no audio from the HDMI.
I've also tried disconnecting the HDMI cable from the RGBtoHDMI device then re-connecting it while it's playing one of the test sounds with no change.

IanSB, 2025-06-22:
Can you try this firmware:

[kernelrpi.zip](https://github.com/user-attachments/files/20853190/kernelrpi.zip)

Put Beta66 or Alpha67A on the SD card.
Unzip the above to get Kernelrpi.img and overwrite the same file on the SD card then reboot.

This changes the frame preamble bits to match the linux driver.

Try both the WAV player and the audio capture.

AWalGH, 2025-06-22:
> Can you try this firmware:
> 
> This changes the frame preamble bits to match the linux driver.
> 
> Try both the WAV player and the audio capture.

The new kernel file loads just fine and is recognized as such in the info menu, unfortunately there does not appear to be any difference in the way it's detected and the scaler still shows no audio from the RGBtoHDMI, regardless of mode. Unplugging and replugging while the device is playing a test file or tone didn't make any difference either.

Just like with the previous firmwares, plugging the device directly into literally any other device I happen to have nearby appears tp work just fine. The audio can also be stopped, changed and toggled as well without any fuss on these other things.