IanSB/RGBtoHDMI #50: Amiga 600 audio

kirkegaard, 2025-07-05:
Im struggling a bit trying to figure out how to pick up stereo signal from my A600. Analogue audio was never my thing :) 
If i understand correctly the RCA has a switch that needs to be shorted to switch from mono to stereo. So do i solder the signal pads to left and right, and ground to the shortbar and then the two short as soon as i connect the jack to the hdmi audio board? Or does ground go to the ground pins on the RCA?

[image]
[image]

[image]
[image]

kirkegaard, 2025-07-05:
I just found this and i think ive figured it out. R334 and R324 seems to be a good place. Or negative side of the caps on the front side of the board, although i think everybody knows how hard that is :D 

[image]

IanSB, 2025-07-06:
I don't have an A600 so have not tried this but I understand that you connect the left and right audio to the centre pins of the left and right audio outputs which might only be accessible from the underside of the PCB then insert unconnected RCA/Phono connectors into both outputs which will cause the switches that short the left and right outputs together to be opened.